boldport/pcbmode #2: refdef "location" errors after new gerber fix

dewib, 2015-04-26:
Not a big issue and simple to 'user fix' but a few of my layouts have acquired unintentional changes to the refdef location after inkscape tweaking and extraction. Some of the extracted values are exponentials and crash the fab process. The following component silkscreen should reproduce the error.

  "silkscreen": {
        "refdef": {
          "location": [
            0.0,
            -9.999999992515995e-07
          ],
          "show": true
        },
        "shapes": {
          "show": true
        }
      }

dewib, 2015-04-27:
Further background... These minor movements in refdef location seem to get picked up when doing Inkscape rotations.

saardrimer, 2015-05-29:
The functionality of the 'refdef' silkscreen extraction isn't ideal. If you move the part or rotate it, the extractor will try to figure out the difference. Unfortunately, this doesn't always work as expected. I'd recommend just placing everything, then manually resetting the 'refdef' location, and only then move them, and extract -- that should work as expected.

closed 2015-11-03 by saardrimer