boldport/pcbmode #5: extract.py expects "translate" elements for all svg objects

peterfillmore, 2015-11-02:
Inkscape (v0.91) appears to not add "translate" tags to some elements during rendering.
This causes the "util/extract.py" processing to fail when extracting due to the process tag returning "None" for the "translate" tag.
Failing code is (util/extract.py):
    50         for shape in shapes:
    51             transform_data = utils.parseTransform(shape.get('transform'))

    87             for refdef_text in refdef_texts:
    88                 # Get refdef group location
    89                 transform_data = utils.parseTransform(refdef_text.get('transform'))

So just need to add a check for a "None" value and insert a (0,0) transform to fix.
Or alternatively eliminate the extraction of the "transform" tag when it is not present in the svg.

saardrimer, 2015-11-02:
Yes, I've run into that, and I thought that I fixed it at parseTransform(). I'll take a look.

saardrimer, 2015-11-02:
In branch 'release4' it's fixed like this in utils.parseTransform():

    if transform == None:
        data['type'] = 'translate'
        data['location'] = Point()

saardrimer, 2015-11-03:
As a side note, if you're starting out with PCBmodE, I'd go with branch 'release4'. It's complete except that I want to do more work with it before officially merging it with master.

closed 2015-11-03 by saardrimer

peterfillmore, 2015-11-03:
Great - looks good. will clone that branch and start kicking the tyres (will write a blog post on laying out the board i'm making since i found it confusing to start - but once I got the process down it clicks together well).

saardrimer, 2015-11-03:
Very pleased to hear that you persist! r4 now supports any number of layers and moving objects is more intuitive. Also, placing additional shapes is done just like components. Take a look at 'buggy' or 'release4-example' for hints. Documentation is lacking ;)

Feel free to get in touch if you run into trouble and I look forward to see what you've created!