boldport/pcbmode #10: Use YAML instead of (or in addition to) JSON?

tommay, 2016-01-12:
I've looked through some of the example JSON files and as usual I find JSON is a lot easier for computers to deal with than it is for people to read and edit.  Have you considered using YAML instead?  Or supporting it in addition?   Functionally it's like JSON but syntactically it's like Python, with indentation and no curly braces or unnecessary quotes to get in the way.  I hand-translated a few files and they got more compact and a lot easier for me to wrap my head around.

saardrimer, 2016-01-13:
I don't have strong feelings about JSON vs YAML. Supporting both shouldn't be too difficult, either when reading the files (YAML-> JSON first) or when assigning file data to dictionaries. But, to be honest, that would be pretty low on my (own) priority list of functionality additions.

tommay, 2016-01-13:
Saar, thanks for getting back to me.  I brought this up because I'm intrigued by PCBmodE but the JSON files are just not beautiful.  I'm a ruby guy who hasn't done python for years but I think I'll give it a shot because YAML would make PCBmodE a lot more accessible for me.

raek, 2016-10-05:
I also prefer editing data in YAML format over JSON if have to do the editing by hand.

I investigated this a bit. There is a neat library called PyYAML which is all-Python, so it is easy to install. With it I made a small tool called "reformat_yaml.py" in this gist: https://gist.github.com/raek/be9a1d9d095ce2462d8df36729ce768b

I wanted to see how the YAML would look after PCBmodE would put it back in the file. Luckily it behaves well and sorts the keys etc. PyYAML supports two visual styles ("block" and "flow") for the innermost arrays and objects, so made a switch in the tool to show the difference. Note that JSON happens to b valid YAML, so I could feed the original JSON files for The Cuttle project into the tool.

The "flow" style:

    # python reformat_yaml.py --mode flow thecuttle.json thecuttle_flow.yaml
    
    components:
      C1:
        footprint: th_2p54mm_2pin
        layer: top
        location: [27.496904, -4.7103649]
        rotate: 90
        silkscreen:
          refdef: {show: false}

The "block" style:

    # python reformat_yaml.py --mode block thecuttle.json thecuttle_block.yaml
    
    components:
      C1:
        footprint: th_2p54mm_2pin
        layer: top
        location:
        - 27.496904
        - -4.7103649
        rotate: 90
        silkscreen:
          refdef:
            show: false
    ...

@tommay: Which one do you like the most?

raek, 2016-10-05:
The original JSON for comparison:

    {
      "components": {
        "C1": {
          "footprint": "th_2p54mm_2pin", 
          "layer": "top", 
          "location": [
            27.496904, 
            -4.7103649
          ], 
          "rotate": 90, 
          "silkscreen": {
            "refdef": {
              "show": false
            }
          }
        }, 
    ...

Visually there's not much differency. The real benefit as I see it is the lower number of characters you need to type and the sensitivity to details such as trailing commas.

tommay, 2016-10-05:
> be9a1d9d095ce2462d8df36729ce768b

Nice!

I have to admit that one of the reasons I haven't actually tried pcbmode is
that I couldn't get past the hand-editing of JSON, especially after
translating a few of the example JSON files to YAML and seeing how much
more manageable and friendly they got.

Not that my opinion matters here, but "flow" does a better job with the
short "location" list, but "block" is better for the small "refdef" hash,
and overall my (unimportant) preference would be block because it's got a
consistent style that's easy to adapt to and maintain when working by hand.

Tom

saardrimer, 2019-12-12:
I'm starting to work on the next version of PCBmodE at branch cinco-master.

I'll re-evaluate this. It seems to me that since PCBmodE reads in JSON files into a dict and would do the same for YAML, supporting both isn't a major task... then people could choose to so what they prefer.