boldport/pcbmode #11: Vias only have pads on top layer

pingswept, 2016-02-26:
(Hi Saar!)

The rocket board I was working on recently (https://github.com/rascalmicro/rocketdmx) had a problem where the vias only had pads on the top layer. The manufacturer caught the problem, so it was no big deal, but I'm trying to fix it for the next revision.

Here's what I've got for via.json (which, other than whitespace, is identical to, for example, https://github.com/boldport/cordwood/blob/master/components/via.json):
```
{
"pins":
  {
  "VIA":
    {	
      "layout": 
      {
        "pad": "via_0.5", 
        "location": [0, 0]
      }
    }
  },
  "pads":
  {
    "via_0.5":
    {
      "shapes":
      [
      	{
          "type": "circle",
      	  "layer": ["top", "bottom"],
      	  "style": "fill",
                "diameter": 0.7,
                "buffer-to-pour": 0.2, 
                "soldermask": {}
        }
      ],
      "drills":
      [
        {
          "diameter": 0.35
        }
      ]
    }
  }
}
```

That seems like it should create vias on top and bottom, but they're still only appearing on top. Might have something to do with the layer mentioned in the via section of <design_name>_routing.json?

saardrimer, 2016-02-26:
`   "layer": ["top", "bottom"],`

needs to be

`   "layers": ["top", "bottom"],`

Otherwise PCBmodE doesn't pick this up (it looks for 'layers', not 'layer') and assumes only 'top' by default. Not obvious, but now you know ;)

pingswept, 2016-02-26:
Ah, that's totally it! Thanks!

The bad syntax appears in your cordwood, cordwood-2ed, and hello-solder repos (but not release4-example). Might be worth fixing, but maybe not. Anyway, thanks!

closed 2016-02-26 by pingswept

saardrimer, 2016-02-26:
Yeah, vias are not used on the cordwood designs... so I never caught it and they lingered in the repo.

saardrimer, 2016-02-26:
All fixed now ;) Thanks!