boldport/pcbmode #17: inkscape extension

Neon22, 2016-05-10:
Hi Saar, I'm making some inkscape extensions.
Do you have a list of things you think would be useful to do in an inkscape extension ?

saardrimer, 2016-05-10:
I'll need to think about it, but don't wait for me. Start and we'll see how it goes. First thing that comes to mind, though, are pre-set layer views... say, 'just top', or 'just assembly', etc.

saardrimer, 2019-12-12:
I'm starting to work on the next version of PCBmodE at branch cinco-master.

Now that Inkscape 1.0 has revamped extensions I think a helper extension could be a good addition. It won't add any functionality, so it won't required, but allow invoking PCBmodE from within Inkscape instead of the command-line.

Neon22, 2019-12-12:
I'm trying to get involved in eth inkscape side of this and would still like to help you with UI inside inkscape.