boldport/pcbmode #20: from . import config (and others) give the error "ValueError: Attempted relative import in non-package"

rickarddahlstrand, 2016-08-17:
When running pcbmode.py on Mac OS I get an error that seems related to the way the includes are written. It seems to be related to this post: http://stackoverflow.com/questions/1112618/import-python-package-from-local-directory-into-interpreter

Any ideas on how to fix this?

```
# python pcbmode.py --help
Traceback (most recent call last):
  File "pcbmode.py", line 15, in <module>
    from . import config
ValueError: Attempted relative import in non-package

```

prospero78, 2016-09-01:
> python-package-from-local-directory-into-interpreter

ValueError: Attempted *relative import in* *non-package*

raek, 2016-10-02:
Newcomer here. I think the way to install and run pcbmode is in the middle of a change. It looks like the code has changed but not the documentation. See pull request #12.

You are not the only one who was bitten by this. I had the same problem, and so did the person in issue #18. I think that to use pcbmode you are now supposed to run `python setup.py` to install pcbmode on the computer and then run it using the `pcbmode` command instead of running Python with the path to `pcbmode.py`.

I created issue #21 to collect up to date information and update the docs.

raek, 2016-11-01:
This should no longer happen with the most recent instructions and code. @rickarddahlstrand can you confirm this?

closed 2019-12-12 by saardrimer