boldport/pcbmode #22: Default pcbmode_config.json file not installed

raek, 2016-10-02:
When running pcbmode from the current master, I cannot get `pcbmode` to find the default `pcbmode_config.json` file.

How to reproduce:

    sudo apt-get install libxml2-dev libxslt1-dev
    virtualenv pcbmode-env
    git clone git@github.com:boldport/pcbmode.git
    source pcbmode-env/bin/activate
    cd pcbmode
    python setup.py install
    cd
    pcbmode -b foo -m

Error message:

    -- Processing PCBmodE's configuration file
    ('pcbmode.pcbmode', 'pcbmode_config.json')
    -----------------------------
    Yikes, ERROR!
    * Couldn't open PCBmodE's configuration file pcbmode_config.json. Looked for it here:
      /home/raek/pcbmode_config.json 
    
    Solder on!
    -----------------------------
    Traceback (most recent call last):
      File "/home/raek/code/pcbmode-env/bin/pcbmode", line 9, in <module>
        load_entry_point('pcbmode==4.1.dev26-ng385e93b.d20161002', 'console_scripts', 'pcbmode')()
      File "build/bdist.linux-x86_64/egg/pcbmode/pcbmode.py", line 356, in main
      File "build/bdist.linux-x86_64/egg/pcbmode/pcbmode.py", line 135, in makeConfig
      File "build/bdist.linux-x86_64/egg/pcbmode/utils/messages.py", line 46, in error
    Exception

raek, 2016-10-02:
The `pcbmode-4.1.dev26_ng385e93b-py2.7.egg` file does not contain the `pcbmode_config.json` file for me. If I add a line in setup.py to include it in the package, then it works.

saardrimer, 2016-10-31:
Is this still an issue on your end? I seem to recall that it got resolved through other changes.

raek, 2016-11-01:
No this was fixed in #23. I can close the issue.

closed 2016-11-01 by raek