boldport/pcbmode #24: Idea: Add command to create a new board

raek, 2016-10-02:
For beginners it would be useful to have a pcbmode command to create a new board directory with .json files all needed sections in place. Perhaps it could contain an outline and an example component as well.

A user shoudl ideally be able to play with the examples in the documentation by just creating a new board with this command and then add the json from the example.

saardrimer, 2016-10-31:
I think that this could be fantastic. At the moment I clone an existing board and manually change the names. It works, but it's a process that could be nicely done... perhaps guided.

saardrimer, 2019-12-12:
I propose adding a switch:

`--new-board <board name>`

that will launch a command-line wizard for creating a new project from a template that's part of the repository. This template will contain a simple design with one or two components and shapes.