boldport/pcbmode #35: Link to boards repositary in tutorial is broken

KidIccurus, 2017-02-08:
Link leads to a Github 404 page. On this page http://pcbmode.readthedocs.io/en/latest/hello-solder.html

JarrettR, 2017-02-08:
Same as issue #29 and #14

KidIccurus, 2017-02-08:
https://github.com/boldport/hello-solder looks like it may work?

KidIccurus, 2017-02-08:
And sorry for duplicating, this is my first time using Github.

KidIccurus, 2017-02-08:
Well, I tried downloading from there but I got an error making the board. With the command
 `pcbmode -b hello-solder -m
`
I Got the flowing output
 `-- Processing PCBmodE's configuration file
-- Processing board's configuration file
-- Creating board
Traceback (most recent call last):
  File "/home/sam/pcbmode/pcbmode-env/bin/pcbmode", line 11, in <module>
    load_entry_point('pcbmode==4.1.dev41+ngba62f96', 'console_scripts', 'pcbmode')()
  File "build/bdist.linux-x86_64/egg/pcbmode/pcbmode.py", line 387, in main
  File "build/bdist.linux-x86_64/egg/pcbmode/utils/board.py", line 21, in __init__
  File "build/bdist.linux-x86_64/egg/pcbmode/utils/module.py", line 58, in __init__
  File "build/bdist.linux-x86_64/egg/pcbmode/utils/module.py", line 965, in _getComponents
AttributeError: 'list' object has no attribute 'get'
`

This is on Linux mint, and I have not yet made anything successfully with this installation.

JarrettR, 2017-02-08:
Yeah, hello-solder was for an older version of PCBModE and isn't compatible.

Try out the tutorial using thelady instead, it should work:

https://github.com/boldport/thelady

KidIccurus, 2017-02-08:
Thank you very much, I will give it a go. Also, should I close this issue since it is a repetition? I don't really know the etiquette to use here? Thanks.

JarrettR, 2017-02-08:
Yeah, closing would keep the issue list cleaner :)

KidIccurus, 2017-02-08:
Just tried the lady. It's running now and appears to have worked. Thank you very much for you help, I will close it now. Sorry about the duplication.

closed 2017-02-08 by KidIccurus

msbeck, 2018-04-19:
I have tried thelady, and I get a successful build, but I do not get an SVG as the tutorial seems to indicate I should. Only another .json "paths_db.json