boldport/pcbmode #49: Autorouting

Immortalin, 2018-05-02:
If one were to develop a custom autorouter it use an external autorouter, what's the best way to integrate with this?

JarrettR, 2018-05-02:
Because routing is done (usually) purely in Inkscape, it might be best to write a standalone Inkscape plugin that works how you like it.

Neon22, 2018-05-02:
I have some experience with Inkscape extensions - this sounds great to me. I'd be happy to help out @Immortalin Do you have any autorouters in mind ? are any of them in python or have python apis ?

Neon22, 2018-05-03:
I found:
- https://github.com/vygr/Python-PCB
- http://razencad.com/
- https://gist.github.com/nmz787/10c60e76941a8e5de624454666ea65b3
   - Monosat (followiing those rabbits sigh) https://github.com/sambayless/monosat

Immortalin, 2018-05-05:
@neon I was thinking of as a general platform for developing autorouters, it's easier with JSON as you won't have to write custom parsers for the proprietary netlists formats (which usually look like a lisp variant)

Neon22, 2018-05-05:
yes json sounds fine - but if you use (see initial post) custom autorouters then you probably need some readers. 
However - where do we begin ?

closed 2019-12-12 by saardrimer