boldport/pcbmode #52: Specify python version for virtualenv

WayneKeenan, 2018-10-13:
Current mainline is python2.

The docs could make it explicit when creating the virtualenv on a host that is defaulting to python3:

virtualenv -p python2 pcbmode-env

JarrettR, 2018-10-13:
Use the fork:

https://github.com/threebytesfull/pcbmode/tree/improve-test-coverage

It runs on Python 3, too

WayneKeenan, 2018-10-13:
That’s cool. Thanks

Just the docs need to reflect the main point of entry for n00bs, don’t they?

JarrettR, 2018-10-13:
I would very much like @saardrimer to merge that branch into mainline!

saardrimer, 2019-12-12:
I'm starting to work on the next version of PCBmodE at branch `cinco-master`.

I understand that for Python 3.5+ we should be using the native `venv`, so I'll update the docs to reflect that. See also #48 .