boldport/pcbmode #54: Estimated first time run on a Pi

WayneKeenan, 2018-10-13:
Just to capture the info for documentation purposes ...

Running the initial generation of an SVG for the ‘binco’ board on a Pi 3 takes in the region of 1.5 to 2 hours.

As there is an initial onetime setup hit I will report back again on a timed run of a secondary invocation.

saardrimer, 2019-12-12:
How does it for after two invocations? ;)

WayneKeenan, 2020-02-09:
Sorry I missed this and sorry but I think that Pi is MIA, unless I ran this on my MythicBeasts one that is still around.

WayneKeenan, 2020-02-10:
I still had the same environment in the cloud... the results for second run were:

```
real	1m6.547s
user	0m58.460s
sys	0m3.904s
```

WayneKeenan, 2020-02-10:
I seem to recall it was rendering the fonts/text that slowed things down a lot, not 100%, been a while.