boldport/pcbmode #62: Detect polarity of SVG paths

saardrimer, 2019-12-12:
At the moment the designer needs to specify the `gerber-lp` polarity sequence of overlapping SVG paths. This is because PCBmodE doesn't know how to detect overlapping paths and figure out the 'holes' (that's because I'm not smart enough to write the code ;)

This function would really improve the usability of PCBmodE and clear out a major pain for beginners and everyone when processing complex shapes.

Neon22, 2019-12-12:
Take a look in this inkscape extension which takes arbitrary inkscape svg and breaks it down to lines for exporting to openSCAD. One of the things it successfully does is deal with holes. I think this might address the polarity (direction?) of svg boundaries.
of course its not in inkscape 1.0 syntax and lots of the functions have changed names.
One day I'll adjust and put in git so it can be maintained but for now you might find it useful ??

https://www.thingiverse.com/thing:1065500

saardrimer, 2019-12-16:
Found this from @mmccoo:
https://nerd.mmccoo.com/2018/01/11/understanding-and-parsing-svg-paths/

Neon22, 2019-12-16:
nice link. It outlines the same approach as used in the thingiverse extension but doesn't use any of the inkscape helper code and doesn't deal with the other primitives like arcs and curves. IMHO either would make a good starting point if you were not able to include them.