boldport/pcbmode #64: Change license to GPL3

saardrimer, 2019-12-12:
Currently the code is available as an MIT license. I'm thinking of moving it to GPL3 before doing the bulk of the work. Thoughts?

Neon22, 2019-12-12:
I think the decision lies with you. If you decide to release it under both licenses that's also possible. The main diff IMHO is that MIT allows for commercial products to use the code.

saardrimer, 2019-12-22:
I've decided to move to GPL 3 starting with Cinco. 

The commonly raised downside of GPL is that commercial companies are concerned about including GPL'd work in their own codebase because it might mean that they may also be required to release it under GPL. This is a valid concern, of course, but not really relevant to PCBmodE as it is not a library that is integrated into other software. Commercial companies can still *use* PCBmodE to create PCBs as the product of the software can be licensed as they wish.

The advantage of GPL is that it prevents -- to the extent that a license can -- commercial use of the software without contributing back. I don't believe in emotions coming into licensing. If a license allows something, there is no 'fair' or 'nice' or 'bad' if someone actually complies with it. The license is the license. So if I don't want others to take the code and commercialise it without releasing their contributions -- and I do not -- GPL is my only practical choice (I wish that there were more).

Finally, KiCad is GPL licensed, and so it re-inforced my choice. 

This is the second license change of PCBmodE! Versions 1 to 3 are Apache-licensed. Version 4 is MIT licensed. Those licenses will continue to apply to these versions. Version 5 (Cinco) will be licensed under GPL 3.

closed 2020-01-04 by saardrimer