boldport/pcbmode #73: Add JSON option to select coordinate origin of object

JarrettR, 2020-01-05:
While drawing shapes in Inkscape and copying the SVG paths into JSON files, the most jarring / time-consuming portion of the process is translating from Inkscape's coordinate system to PCBModE's system.

That is, Inkscape object have an origin at the bottom left of the screen, while PCBModE's is in the centre. So copying Inkscape coordinates to a JSON file and then using that to generate a new SVG, and opening it up in Inkscape results in an incorrectly placed object.

Allowing an additional parameter in the JSON file would be helpful, say:
```

        {
          "type": "path",
          "layers": ["top"],
          "value": "m -1.1632,305.44917 0.76716,0.0716 -0.008,0.0852 q -0.0326,-6.2e-4 -0.0943,-0.002 -0.0617,-9.1e-4 -0.0878,0.006 -0.0411,0.0119 -0.0616,0.0366 -0.0206,0.0259 -0.0251,0.074 l -0.11521,1.23418 q -0.004,0.042 0.007,0.0709  -0.0961,-0.0211 z",
          "style": "fill",
          "location": [
            -0.88,
            -9.52
          ],
          "origin": "bottom-left",
          "rotate": 0
        },
```

This would simply take the given coordinates and subtract the (bounding-box / 2) in both the X and Y directions.

This parameter could be used for shapes or components, even vias I guess, although that would be strange.

saardrimer, 2020-01-05:
If it's a component or shape (json) with a placement marker then it's not too bad:

1. Place component/shape with some rough coordinates
2. Generate SVG
3. Move marker to desired position
4. Extract

This applies to vias as well, which you can cut-and-paste, then extract.

But if there are multiple shapes in one list of shapes then getting their relative position -- which is what you're pointing out I suspect -- is more of a pain and is a bit of a trial-and-error unless you measure in Inkscape... and that's not too straightforward either.

My thinking was having a global setting that lets the user choose their board's origin instead of on an individual shape basis.

I'd like to solve the above problem. Another option is for PCBmodE to extract the relative positions in addition to the placement markers... or have a marker per-shape rather than per-component.

Need to think more on how to do that as I develop 'cinco-master', the next version.

JarrettR, 2020-01-05:
Cinco is definitely why I'm mentioning it now :)

The reason I suggest the new `origin` option is that it basically eliminates that entire 4-step workflow you posted, for a certain subset set of problems.

If I want to create a circular array of shapes, I can easily enough generate a list of coordinates that they need to be placed at - it takes 5 minutes to copy them all into JSON. But if I also need to figure out some offsets, or attempt to position them by hand (inaccurately) and do some trial-and-error to make it look right, it can easily turn into an hour-long job.

The nice thing about doing it locally per-placement instead of globally, is when you do your usual code->generate SVG->tweak->extract workflow, the local `origin` option will disappear on its own.

Here's what I mean:

1. Place component/shape with some exact coordinates, with the `origin` set to bottom-left
2. Generate SVG. PCBmodE translates the coordinates internally. There is no `origin` option stored in the SVG.
3. Tweak other components on the SVG if desired
4. Extract. The shape's coordinates have been brought back into JSON, but with the actual, centred-origin coordinates that is normal with this workflow. There is no `origin` option in this JSON file anymore.

It's definitely a minor addition, but I feel it has fairly profound impacts for user-friendliness